aromring/MAX30102_by_RF #1: Sensor not turning on

RimonB95, 2018-04-29:
I copied over the max30102 cpp and h file form your project into the original project that i download from maxim website and removed the softi2cmaster library.
but each time i run the code it gets stuck in this line 
while(digitalRead(10)==1);
i can never pass this line and the sensor doesn't even turn as it doesn't light up red
any idea what can i do?
i'm trying to use the code the ESP8266 12E (nodeMCU 1.0 board) but i'm trying it on the arduino UNO first to make sure it is working.
Thanks

aromring, 2018-04-30:
Hi RimonB95,
Thank you for this report. If I understand correctly, you have cherry-picked only two files from my repository, max30102.cpp/h, and incorporated these into MAXREFDES117# code from MAXIM, is this correct? Another tiny bit of information you provided is this line of code 
while(digitalRead(10)==1);
your project gets stuck at - it's in the MAXIM's code.
You know, I always got very irritated after calling a tech support at some corporation and instead of having them dive deep into the matter of the problem they first responded with obvious BS, like "make sure your device is turned on" - duh!! But now I have a new understanding of why this happened - you provided such a tiny amount of information that I am forced to ask obvious questions first - I am sorry: 1) Did you check your connections? Pin 10 should be tied to pin INT of the MAX30102. Check my Instructable on how I connected everything:
http://www.instructables.com/id/Pulse-Oximeter-With-Much-Improved-Precision/
2) Did you test my code in its entirety? I am not a psychic and have no idea how you implemented the code merger - something might not match.
Again, I am sorry for this "duh", but if you want to save time and have me provide a more meaningful help right away, then you have to provide more information: like schematic of connections and what exactly you did to the code.

aromring, 2018-04-30:
A few more "duh" questions:
3) Are you using the MAXREFDES117# System Board or bare MAX30102 chip by itself? The latter has much more strict voltage requirements, you know.
4) Did you try MAXIM's code as originally posted, _without_ my modifications? Did it work?

aromring, 2018-06-15:
I haven't heard from you in a long time. Hence, I consider this issue to be closed.

closed 2018-06-15 by aromring

ElectronicsInnovation, 2021-03-18:
This video might help you, in this video, an Indian guy explained the issue in detail with the help of the max30100 sensor schematic diagram, and solved it in two ways. https://youtu.be/ZqdmA4NAqb0