aromring/MAX30102_by_RF #3: Unable to print SPO2 and HR via serial connection

jaredwong92, 2018-06-27:
I am integrating MAX30102 with Arduino without Adafruit's Feather M0 Adalogger. Although I changed the output to Serial port (at baud rate 115200), the SPO2 and HR numbers are zero. Below is the example of the output from Serial pot. I had attached the code as well. 

Start calculate heart rate
--RF--
529	0.00	0	0:8:49
------
Start calculate heart rate
--RF--
530	0.00	0	0:8:50

[RD117_ARDUINO_new1.zip](https://github.com/aromring/MAX30102_by_RF/files/2141667/RD117_ARDUINO_new1.zip)

aromring, 2018-06-28:
Well, no wonder the SpO2 and HR numbers are zero: in your code you have commented out rf_heart_rate_and_oxygen_saturation() - the very routine that does all the signal processing!
Here, try this code and let me know of the outcome. Please copy and paste _full_ output in the Serial monitor.

[RD117_ARDUINO_new2.zip](https://github.com/aromring/MAX30102_by_RF/files/2143347/RD117_ARDUINO_new2.zip)

jaredwong92, 2018-06-28:
Previously I tried to use the rf_heart_rate_and_oxygen_saturation() routine but it will not run pass that routine. That is why I commented out. Below is the full output in the Serial Monitor

Vbatt=	1.62
blink LED
Initialise MAX30102
Start calculate heart rate
Vbatt=	0.19
blink LED
Initialise MAX30102
Start calculate heart rate

aromring, 2018-07-01:
All right, it's time to see the raw data coming from the sensor. I need it to debug this issue. Please run the attached code and report _full_ output in the Serial monitor.

[RD117_ARDUINO_new3.zip](https://github.com/aromring/MAX30102_by_RF/files/2152359/RD117_ARDUINO_new3.zip)

jaredwong92, 2018-07-02:
This is the full output in the Serial monitor.
[output.txt](https://github.com/aromring/MAX30102_by_RF/files/2155961/output.txt)

aromring, 2018-07-03:
Ah, we are getting somewhere. It looks like your microcontroller is reaching rf_heart_rate_and_oxygen_saturation(), can't handle it, then quits and restarts. This is why it runs setup() repetitively. My guess is that it runs out of memory.
Thus: what microcontroller do you use? When you compile the code (with Arduino IDE, I presume) you get the report of memory usage: what does it say about % memory used by code and data, respectively?

Adalogger's Cortex M0 microcontroller has enough memory and computational power to handle my algorithm. Yours may not...

boblongoria, 2018-07-04:
I admire your heart or service for the rookie problems of jaredwong92.
I will attempt to learn from his mistakes and also benefit from your analisis.
Regards,
Bob

jaredwong92, 2018-07-05:
I am using ATmega32U4. You are right that the memory usage is very high. 
Sketch uses 15160 bytes (46%) of program storage space. Maximum is 32256 bytes.
Global variables use 1902 bytes (92%) of dynamic memory, leaving 146 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur. 
Guess I need to use another algorithm to get the HR and SPO2

aromring, 2018-08-23:
Jared,
I suggest a better MCU; apparently ATmega32U4 is insufficient.
Unless you have other issues I consider this issue to be closed.

closed 2018-08-23 by aromring