aromring/MAX30102_by_RF #4: IR and Red signals do not show any heart rate info

doug-burrell, 2019-04-30:
Hi, I've ported your code to Python to work on a Raspberry Pi and it works great on some sample data I found in another repository but when I run it the IR and Red signals are relatively flat with no apparently heart rate signal. These signals do change when I move my finger around so I know the sensor is working but any changes due to heart rate are not present. Either I have no pulse or I haven't configured the MAX30102 correctly. Can you suggest what parameters in the sensor's config may need to modified to get a valid heart rate?
Thanks!

aromring, 2019-05-01:
Well, without additional info all I can do is to provide a (useless) suggestion that your either your sensor is bad, or some connections are incorrect.
Please go to [README](https://github.com/aromring/MAX30102_by_RF) and scroll down to HOW TO REPORT BUGS section for a more useful info.

doug-burrell, 2019-05-01:
I don't believe it's a bug in your code but just wondering if during your time working with the max30102 you saw this same situation.

aromring, 2019-05-01:
Yes, I did see this happening when I mistakenly connected the INT port to a wrong MCU pin. :) 
So here is another suggestion: buy or borrow the M0 Adalogger (or an equivalent MCU), download free Arduino IDE, and prove to yourself that all the hardware and connections work properly. Then you will be able to debug your Python code on Raspberry Pi. Is it interrupts? Maybe the I2C communication is faulty? I have never used Raspberry, thus I have no idea how this platform handles I2C.

doug-burrell, 2019-05-01:
Appreciate the feedback about the interrupt pin. I've modified my setup to not require the interrupt pin, instead it reads the FIFO read and write pointers to determine if data is available. But it's a good idea to try to completely copy your hardware setup, I will see about getting an M0 adalogger.

Did you ever encounter issues with cold fingers not producing proper red and ir signals?

aromring, 2019-05-01:
No, I have not noticed any issues tied to finger's temperature. But, on the other hand, it's quite difficult to find extreme cold here in California. :)

closed 2019-05-01 by aromring

doug-burrell, 2019-05-01:
Ok, just wondering. Thanks for your help!