aromring/MAX30102_by_RF #5: Are we confident enough in the data coming from the board?

kwende, 2019-05-15:
Greetings - 

I was trying to find your email so that I could ask this directly, but I cannot find it so I thought I'd open an issue. I'm not reporting a bug for your software, but mostly curious about an observation I've had when using the board, and also when observing the data plots. 

Namely this: the flat line for heart rate and SPO2 seems just plain wrong. The heart rate varies quite a bit during sleep, in fact, there should be a "saddle" effect that occurs when you're sleeping properly. Also, during REM sleep, breathing rate and heart rate also increase. In the data plot you're showing, it's a straight line with little variability. 

I have a belief I might have sleep apnea, so I bought one of these boards. I used the "noisy" version of the software provided by the manufacturer but noticed my SPO2 was constant throughout the entire night at nearly 97. This was suspect for me. 

Do we know for sure the data coming off these boards are even right? Both oxygen and heart rate should vary more than what I see when observing my data plots as well as that of others. I thus have a great deal of suspicion about how good these boards even function. 

My data plot: 

[image]

boblongoria, 2019-05-15:
I am also interested in this issue but lacking in technical expertise. Regards Bob

[image]<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F526471%2F57781277-77c26280-76ef-11e9-8579-868e89b9dde3.png&data=02%7C01%7C%7Ca20eb635b8a3480d33fc08d6d93d4be1%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636935254613898752&sdata=SNi4Wg5OWzbeNL%2BfF%2FZHZNz87RKw7eHlhXQdnjGeEW8%3D&reserved=0>

kwende, 2019-05-15:
I've since realized the author is communicating with people on his Instructables article. I've closed this comment.

closed 2019-05-15 by kwende

aromring, 2019-05-17:
I've posted the answer in my Instructable.