aromring/MAX30102_by_RF #8: Heart-rate is always the same (83 Bpm)

asemchenko, 2020-03-28:
Hi!

I have used this library with Arduino Nano, and it looks like SpO2 is calculated right but heart-rate value is always 83 and does not changes. Can you help me to understand what is going wrong and how to deal with it, please? 
[image]

I am using the sketch below (there is no any changes, I've just commented some features (DEBUG, ADALOGGER, SAVE_ROW_DATA)
```
/********************************************************
*
* Project: MAXREFDES117#
* Filename: RD117_ARDUINO.ino
* Description: This module contains the Main application for the MAXREFDES117 example program.
*
* Revision History:
*\n 1-18-2016 Rev 01.00 GL Initial release.
*\n 12-22-2017 Rev 02.00 Significantlly modified by Robert Fraczkiewicz
*\n 08-22-2018 Rev 02.01 Added conditional compilation of the code related to ADALOGGER SD card operations
*
* --------------------------------------------------------------------
*
* This code follows the following naming conventions:
*
* char              ch_pmod_value
* char (array)      s_pmod_s_string[16]
* float             f_pmod_value
* int32_t           n_pmod_value
* int32_t (array)   an_pmod_value[16]
* int16_t           w_pmod_value
* int16_t (array)   aw_pmod_value[16]
* uint16_t          uw_pmod_value
* uint16_t (array)  auw_pmod_value[16]
* uint8_t           uch_pmod_value
* uint8_t (array)   auch_pmod_buffer[16]
* uint32_t          un_pmod_value
* int32_t *         pn_pmod_value
*
* ------------------------------------------------------------------------- */

#include "Arduino.h"
#include <Wire.h>
#include <SPI.h>
#include "algorithm_by_RF.h"
#include "max30102.h"

//#define DEBUG // Uncomment for debug output to the Serial stream
//#define USE_ADALOGGER // Comment out if you don't have ADALOGGER itself but your MCU still can handle this code
//#define TEST_MAXIM_ALGORITHM // Uncomment if you want to include results returned by the original MAXIM algorithm
//#define SAVE_RAW_DATA // Uncomment if you want raw data coming out of the sensor saved to SD card. Red signal first, IR second.

#ifdef USE_ADALOGGER
  #include <SD.h>
#endif

#ifdef TEST_MAXIM_ALGORITHM
  #include "algorithm.h" 
#endif

// Interrupt pin
const byte oxiInt = 10; // pin connected to MAX30102 INT

// ADALOGGER pins
#ifdef USE_ADALOGGER
  File dataFile;
  const byte chipSelect = 4;
  const byte cardDetect = 7;
  const byte batteryPin = 9;
  const byte ledPin = 13; // Red LED on ADALOGGER
  const byte sdIndicatorPin = 8; // Green LED on ADALOGGER
  bool cardOK;
#endif
[... 308 lines cut ...]
// blink three times if isOK is true, otherwise blink continuously
void blinkLED(const byte led, bool isOK)
{
  byte i;
  if(isOK) {
    for(i=0;i<3;++i) {
      digitalWrite(led,HIGH);
      delay(200);
      digitalWrite(led,LOW);
      delay(200);
    }
  } else {
    while(1) {
      for(i=0;i<2;++i) {
        digitalWrite(led,HIGH);
        delay(50);
        digitalWrite(led,LOW);
        delay(50);
      }
      delay(500);
    }
  }
}
#endif
```

aromring, 2020-03-29:
Hi Andrii,
Yes, I will be glad to help. Please open the README file here https://github.com/aromring/MAX30102_by_RF/blob/master/README.md
scroll down to HOW TO REPORT BUGS section and follow instructions therein.

akuznets0v, 2020-03-30:
Hello, just wanted to chime in here that I ran the above code on an Arduino Uno and it's giving accurate-ish heart rate as compared to a Santa Medical SM-110 @asemchenko

asemchenko, 2020-03-30:
Hello @akuznets0v.
That is interesting. Did you run it with MAX30102 sensor?

akuznets0v, 2020-03-30:
@asemchenko Tested with Arduino Uno R3, MAXREFDES117# from Digikey, and using pin connection setup from Maxim (https://www.maximintegrated.com/en/design/reference-design-center/system-board/6300.html). Note that this is 3.3v.

(arduino pin) (sensor pin)
SDA SDA
SCL SCL
10 INT
3.3V	VIN
GND GND

supersuperfranz, 2020-03-31:
I have the same issue. i'll collect data and sent it to you but i write here if you have yet solved.

aromring, 2020-03-31:
@supersuperfranz, please do. As long as someone sends me raw data, I can debug the issue. Do you also use Arduino Nano?

supersuperfranz, 2020-03-31:
Yes nano and brand new digikey sensor.
Tomorrow I'll send you data!
I would arrange it in case of needs due to coronna issue here in Italy.

Thanks

supersuperfranz, 2020-04-01:
I've collected some raw data, is enough?
However both Adafruit and Maxim algorithms fails when usigng O2 mode. Only
1 led adafruit beatrate seems to work. Maybe something changed in
configuration registers or fifo buffers? In the last max30102 datasheet i
cannot find the pilot led instructions. is maybe deprecated in new chips?

thanks a lot for your help

Francesco Tripaldi

Press any key to start conversion
Time[s]	SpO2	HR	Clock	Ratio	Corr	0	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16	17	18	19	20	21	22	23	24	25	26	27	28	29	30	31	32	33	34	35	36	37	38	39	40	41	42	43	44	45	46	47	48	49	50	51	52	53	54	55	56	57	58	59	60	61	62	63	64	65	66	67	68	69	70	71	72	73	74	75	76	77	78	79	80	81	82	83	84	85	86	87	88	89	90	91	92	93	94	95	96	97	98	99	0	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16	17	18	19	20	21	22	23	24	25	26	27	28	29	30	31	32	33	34	35	36	37	38	39	40	41	42	43	44	45	46	47	48	49	50	51	52	53	54	55	56	57	58	59	60	61	62	63	64	65	66	67	68	69	70	71	72	73	74	75	76	77	78	79	80	81	82	83	84	85	86	87	88	89	90	91	92	93	94	95	96	97	98	99
1	79.51	83	0:0:1	0.74	1.00	5928	597	597	597	597	597	597	597	597	597	597	597	597	597	11626	600	600	600	600	600	600	600	600	600	600	600	600	600	600	600	600	600	21850	605	605	605	605	605	605	605	605	605	605	605	605	605	605	605	605	605	38766	631	631	631	631	631	631	631	631	631	631	631	631	631	631	631	631	631	78940	650	650	650	650	650	650	650	650	650	650	650	650	650	650	650	650	650	98279	576	576	576	576	576	576	576	576	576	576	576	576	576	6096	634	634	634	634	634	634	634	634	634	634	634	634	634	11291	634	634	634	634	634	634	634	634	634	634	634	634	634	634	634	634	634	19968	636	636	636	636	636	636	636	636	636	636	636	636	636	636	636	636	636	34454	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	69914	738	738	738	738	738	738	738	738	738	738	738	738	738	738	738	738	738	102615	691	691	691	691	691	691	691	691	691	691	691	691	691
1	75.33	83	0:0:1	0.68	0.93	101376	103563	106664	679	679	679	679	679	679	679	679	679	679	603	603	603	603	603	603	603	603	603	603	603	603	603	603	603	603	603	110634	575	575	575	575	575	575	575	575	575	575	575	575	575	575	575	575	575	111065	511	511	511	511	511	511	511	511	511	511	511	511	511	511	511	511	511	111364	593	593	593	593	593	593	593	593	593	593	593	593	593	593	593	593	593	111453	605	605	605	605	605	605	605	605	605	605	605	605	605	605	605	104135	107495	113902	803	803	803	803	803	803	803	803	803	120120	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	733	122698	686	686	686	686	686	686	686	686	686	686	686	686	686	686	686	686	686	123734	638	638	638	638	638	638	638	638	638	638	638	638	638	638	638	638	638	124471	706	706	706	706	706	706	706	706	706	706	706	706	706	706	706	706	706	124785	721	721	721	721	721	721	721	721	721	721	721	721	721	721	721
2	80.00	83	0:0:2	0.81	1.00	111440	645	645	645	645	645	645	645	645	645	111341	649	649	649	649	649	649	649	649	649	649	649	649	649	649	649	649	649	111404	800	800	800	800	800	800	800	800	800	800	800	800	800	800	800	800	800	111411	885	885	885	885	885	885	885	885	885	885	885	885	885	885	885	885	885	111341	860	860	860	860	860	860	860	860	860	860	860	860	860	860	860	860	860	111341	824	824	824	824	824	824	824	824	824	824	824	824	824	824	824	824	824	124631	781	781	781	781	781	781	781	781	781	124462	742	742	742	742	742	742	742	742	742	742	742	742	742	742	742	742	742	124496	918	918	918	918	918	918	918	918	918	918	918	918	918	918	918	918	918	124452	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	1001	124224	986	986	986	986	986	986	986	986	986	986	986	986	986	986	986	986	986	124123	910	910	910	910	910	910	910	910	910	910	910	910	910	910	910	910	910
2	87.18	83	0:0:2	0.52	0.99	111269	111301	111356	2351	2351	2351	2351	111357	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	5928	111323	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	11626	111358	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	21850	111454	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	38766	111508	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	78940	111481	98279	98279	124019	124074	124141	2549	2549	2549	2549	124225	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	6096	124221	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	11291	124245	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	19968	124357	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	34454	124431	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	69914	124370	102615	102615
3	92.33	83	0:0:3	0.87	0.95	111341	111341	111341	111341	111341	111341	111683	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111341	111711	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111269	111718	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111301	111691	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111356	111629	111357	111357	111357	111357	111357	111357	111357	111357	111357	111357	111357	111357	111 [... cut ...] 	114530	114530	114530	114530	114530	114530	114530	114530	115723	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	114496	115782	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	114571	115947	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	114593	115992	114615	114615	114615	114615	114615	114615	114615	114615	11

supersuperfranz, 2020-04-01:
could be related to the  maxim_max30102_read_fifo routine?
Because Wire.requestFrom() is limited to BUFFER_LENGTH which is 32 on the Uno?

akuznets0v, 2020-04-01:
Hello, 

I found an old arduino nano clone (uses ATmega328P Old Bootloader) I had lying around and hooked things up. Looks like everything is stagnant. 

Time[s]	SpO2	HR	Clock	Ratio	Corr
0	80.14	100	0:0:0	1.00	1.00
4	80.14	100	0:0:4	1.00	1.00
9	80.14	100	0:0:9	1.00	1.00
13	80.14	100	0:0:13	1.00	1.00
(and so on)

aromring, 2020-04-02:
@supersuperfranz, thank you so much! Thanks to your effort in collecting raw data coming from the sensor, _as received_ by Arduino Nano, I can address this issue. First, take a look at plots of RED and IR signals plotted from your data; in ArduinoNano*.png files here:

https://www.dropbox.com/sh/5bdi49drt8p5uiu/AADyJRqmlbBmZ1ykzpq_33rva?dl=0

Both signals remain flat except either a sharp dip, or spike occurring every 18 samples (BTW, this interval corresponds to HR = 83 with default settings). Very strange, isn't it? These are a far cry from expected good signals coming from the Maxim sensor; see ExpectedGoodQualitySignals.png. As I wrote in the README, "don't expect a bad signal to produce correct results!"
Why the signal is bad is another matter. My guess is that's a hardware problem caused by an underpowered MCU in Arduino Nano; issue 1) in README. You had a good hunch to look at the maxim_max30102_read_fifo() routine - Maxim has placed this comment in front:
"//Arduino Uno doesn't have enough SRAM to store 100 samples of IR led data and red led data in 32-bit format
//To solve this problem, 16-bit MSB of the sampled data will be truncated.  Samples become 16-bit data."
You may try to replace uint32_t by uint16_t all over the code, but I am skeptical about this hack. Just get a better MCU! As @akuznets0v reported, Arduino Uno works fine.

I am about to close this "83 bpm" issue, unless I hear some new info in the next few days.

akuznets0v, 2020-04-03:
We'll probably want to try these hacks/modifications out and work through a more efficient solution (that will likely work for the original devices). Would you prefer we fork or just start a new branch?

aromring, 2020-04-03:
@akuznets0v, music to my ears! By all means, please investigate how to adopt Arduino Nano to get the right reading from MAX30102. Since I expect code modifications to be minor, with a few #ifdef's for Nano, please start a new branch.

asemchenko, 2020-04-03:
@aromring  Thanks for your answer, I will try with Arduino Uno. But I don't quite understand, how can it happen? As I know, Arduino Nano and Arduino Uno are both based on the same microcontroller (Atmega 328) aren't they?

akuznets0v, 2020-04-03:
To add onto the point raised by @asemchenko, they have the same specs in terms of SRAM and etc.
https://www.elprocus.com/wp-content/uploads/difference-between-arduino-uno-and-arduino-nano.jpg 
(from https://www.elprocus.com/an-overview-of-arduino-nano-board/). This might have to be taken into a different discussion, an initial test would be to see if sampling rarer or switching to uint16_t helps.

aromring, 2020-04-03:
@asemchenko I wish I could give you a meaningful answer, but all Nanos I've ever had were used in my projects long time ago and I've never owned an Uno. I am buying my MCUs almost exclusively from Adafruit...
@akuznets0v I think I owe you a correction. The way GitHub works, only the project owner is able to create branches. All others have to fork, modify their version, then create a pull request to have these changes incorporated.

akuznets0v, 2020-04-03:
aha! I think I have a good lead on the issue. I also received a MAX30102 module from a Chinese firm to compare and it needed a 10K resistor on SDA and SCL to get running so be aware of that. 

I connected my arduino Uno to 3.3v and got readings slowly (once every 5 seconds) that were accurate. I connected it to 5v and I was getting measurements rapidly (once every 1-2 seconds) BUT the bpm was stuck at 83. 

I'll try sending over output from both to @aromring soon.

aromring, 2020-04-05:
Update: under the same link I've posted previously:
https://www.dropbox.com/sh/5bdi49drt8p5uiu/AADyJRqmlbBmZ1ykzpq_33rva?dl=0
I have placed an updated Excel worksheet with new data from @akuznets0v (thank you very much, Andrew!). The data and exemplary plots are placed in the "FromAndrew" spreadsheet. And they are telling a lot. I short, Andrew took raw data from 2 sensors: an original MAXREFDES117# from Digikey and a cheap Chinese clone from Makerfabs under two voltage levels: 3.3 V and 5 V.
Look at the graphs, the differences are startling. The Digikey sensor delivers the expected signals at both voltages with the usual distinction between low and high quality ones as defined by the algorithm settings. The Makerfabs clone, on the other hand, produces weird noise at 3.3 V and strange rectangular waves at 5V. Interestingly, at 5 V some of these waves were semi-periodic (last graph) and the algorithm did accept these and calculated HR = 55 bpm in one case. Soooo.... Does it mean the sensor is working fine at 5V and the problem is communication with the MCU?

akuznets0v, 2020-04-05:
After diving into it, I think this issue may be at fault for the cheaper boards.
https://reedpaper.wordpress.com/2018/08/22/pulse-oximeter-max30100-max30102-how-to-fix-wrong-board/

aromring, 2020-04-05:
@akuznets0v, thank you for this amazing find. It's open source collaboration at its best! :) 
The classic "you reap what you sow" more than adequately summarizes the cheap crap boards that carry the MAX30102 sensor. 
One mystery remains, though. @supersuperfranz wrote that he used "brand new digikey sensor". Hmmm... But I have just checked digikey.com and found two evaluation boards:
1) The Maxim original MAXREFDES117# for $16.34
2) The SEN-15271 one from SparkFun Electronics for $10.95
@supersuperfranz, which one do you have?
BTW, I am not trying to imply anything about the quality of SEN-15271. Its schematics seems fine, the 3.3 V logic is translated into 1.8 V logic via MOSFETs, but the ultimate test is to pair it with a capable MCU and record raw data again.

supersuperfranz, 2020-04-05:
The first one!

I'm waiting samd21 m0 from Amazon for try. I suspect memory overflow, tried
16bit register and less sampling with no results.

akuznets0v, 2020-04-06:
As an update, I implemented the 'cut' and 'add wire' fix and was able to avoid the 83 bpm at 5v. After I got it working I tested the following at 3.3v and 5v
Time[s]	SpO2	HR	Clock	Ratio	Corr
24	99.70	57	0:0:24	0.57	0.96
29	99.74	57	0:0:29	0.73	0.98
38	99.66	62	0:0:38	0.88	0.98

Time[s]	SpO2	HR	Clock	Ratio	Corr
12	99.74	62	0:0:12	0.88	0.98
17	99.80	60	0:0:17	0.86	0.98
30	99.43	65	0:0:30	0.90	0.96

Unfortunately before getting the raw data I tried to check the voltage of the second regulator as recommended by the blogpost and broke something in my attempt. The LED has not turned on since. I have around 20 of these chips so I'll reattempt getting raw data soon.

jointothedarkside, 2020-04-07:
@akuznets0v Did you calibrate your results with something else? are you getting the right value?

akuznets0v, 2020-04-07:
@jointothedarkside I haven't recalibrated and am using the coefficients in the repository. Those bpm results were more or less accurate compared to the Santa Medical SM-110. I was just sitting in my chair when measuring.

According to the makerfabs folks, the voltage across the second regulator is 1.8v for their modules, so it seems just cutting and wiring is required for fixing their 30102 module. I'll be testing those when they arrive. I'll be doing my own fixing and tests in the meantime.

aromring, 2020-04-14:
Since there is no further activity in this thread and the issue turned out to be related to a faulty hardware, I am closing it.

closed 2020-04-14 by aromring