aromring/MAX30102_by_RF #9: Wrong Readings

jointothedarkside, 2020-04-01:
Hello there I read your comment about changing the values if your heart rate is above 60bpm, my problem is that the reading goes down to 51 and up again to 100/107, if I use the regular library the reading is too erratic it can go down to 20 and up to 200, I'm not sure what can be done or how to fix it:
Here's with your algorithm:
Press any key to start conversion
Time[s]	SpO2	HR	Clock	Ratio	Corr
1	90.69	100	0:0:1	0.52	0.86
1	82.68	100	0:0:1	0.56	1.00
3	66.89	100	0:0:3	0.51	0.99
4	42.65	51	0:0:4	0.80	0.99
6	99.58	107	0:0:6	0.55	0.87
8	46.33	100	0:0:8	0.52	1.00
9	58.07	100	0:0:9	0.55	0.99
9	60.20	51	0:0:9	0.65	0.95
11	66.92	100	0:0:11	0.51	1.00
12	65.64	100	0:0:12	0.52	1.00
12	56.33	51	0:0:12	0.64	1.00
15	69.23	51	0:0:15	0.74	1.00
19	76.69	51	0:0:19	0.72	0.96
20	53.66	51	0:0:20	0.91	1.00
21	78.59	51	0:0:21	0.69	1.00
21	80.72	51	0:0:21	0.83	1.00
22	58.30	51	0:0:22	0.76	1.00
22	67.87	51	0:0:22	0.83	1.00
23	67.11	51	0:0:23	0.76	0.99
24	51.70	100	0:0:24	0.51	0.95
25	27.25	51	0:0:25	0.69	0.81
26	99.86	51	0:0:26	0.99	0.92
29	11.86	51	0:0:29	1.01	1.00
29	40.34	51	0:0:29	0.69	0.86
31	58.53	51	0:0:31	0.74	0.88
31	65.91	51	0:0:31	0.91	0.93
32	17.21	100	0:0:32	0.50	1.00
35	79.89	51	0:0:35	0.83	1.00

Here's with the standard algorithm from Maxim

Press any key to start conversion
Time[s]	SpO2	HR	SpO2_MX	HR_MX	Clock	Ratio	Corr
1	-999.00	-999	29.19	107	0:0:1	0.43	0.99
2	-999.00	-999	14.99	107	0:0:2	0.33	0.93
2	-999.00	-999	33.64	107	0:0:2	0.26	0.98
2	-999.00	-999	79.54	51	0:0:2	-0.27	0.93
8	-999.00	-999	75.07	68	0:0:8	-0.14	0.79
8	-999.00	-999	91.89	107	0:0:8	0.09	0.70
9	59.89	107	-999.00	-999	0:0:9	0.51	0.91
9	-999.00	-999	63.20	136	0:0:9	0.51	0.48
9	-999.00	-999	3.49	60	0:0:9	0.07	0.88
11	-999.00	-999	44.12	20	0:0:11	0.47	0.89
11	57.63	51	-999.00	115	0:0:11	0.53	0.84
11	87.51	51	99.95	83	0:0:11	0.77	0.84
15	-999.00	-999	78.31	107	0:0:15	0.12	0.93
15	-999.00	-999	51.86	62	0:0:15	0.17	0.94
17	-999.00	-999	72.32	107	0:0:17	-0.22	0.97
18	7.87	107	-999.00	107	0:0:18	0.50	0.93
18	-999.00	-999	50.92	115	0:0:18	0.50	0.76
21	-999.00	-999	40.03	53	0:0:21	-0.28	0.79
24	-999.00	-999	77.68	32	0:0:24	0.39	0.75
25	-999.00	-999	98.85	125	0:0:25	0.39	0.43
29	-999.00	-999	91.89	60	0:0:29	-0.31	0.88
31	93.04	107	-999.00	-999	0:0:31	0.55	0.95
33	-999.00	-999	33.64	32	0:0:33	0.33	0.89
33	-999.00	-999	94.34	57	0:0:33	0.00	0.83
33	-999.00	-999	81.32	115	0:0:33	0.37	0.94
34	-999.00	-999	94.01	115	0:0:34	0.37	0.77
35	-999.00	-999	29.19	83	0:0:35	-0.05	1.00
35	-999.00	-999	42.10	115	0:0:35	-0.06	0.94
36	-999.00	-999	41.07	83	0:0:36	-0.05	1.00
36	52.17	107	44.12	107	0:0:36	0.53	0.91
36	47.22	107	38.99	83	0:0:36	0.54	0.92
36	-999.00	-999	24.60	107	0:0:36	0.47	0.85
38	-999.00	-999	28.06	57	0:0:38	0.02	0.98
38	-999.00	-999	99.48	107	0:0:38	-0.07	0.91
39	-999.00	-999	87.62	78	0:0:39	0.40	0.94
39	-999.00	-999	77.04	107	0:0:39	-0.07	0.92
40	-999.00	-999	65.60	107	0:0:40	-0.07	1.00

What's happenning :(

My connections are the same but I'm using an ESP32 Board and I'm using the GPIO15 as the INT pin

Thanks

aromring, 2020-04-01:
I will be happy to help. Please open the README page:
https://github.com/aromring/MAX30102_by_RF/blob/master/README.md
scroll down to "HOW TO REPORT BUGS" section and follow instructions therein.

aromring, 2020-04-03:
This behavior looks like another case of the known and (at the moment) open issue #6. Hence, I am closing it as a duplicate, unless I see the requested raw data that convinces me otherwise.

closed 2020-04-03 by aromring