aromring/MAX30102_by_RF #11: 如何解决高心率下的误差。还有实际误差是2倍还是3倍、谢谢！ 如何解决高心率下的误差。还有实际误差是2倍还是3倍、谢谢！ How to solve the error of high heart rate. And the actual error is 2 times or 3 times, thank you

GOKUandVEGET, 2020-04-10:


aromring, 2020-04-10:
I will be glad to help. Please open the README file here: 
https://github.com/aromring/MAX30102_by_RF/blob/master/README.md
scroll down to HOW TO REPORT BUGS section and follow instructions therein.

GOKUandVEGET, 2020-04-11:
> I will be glad to help. Please open the README file here:
> https://github.com/aromring/MAX30102_by_RF/blob/master/README.md
> scroll down to HOW TO REPORT BUGS section and follow instructions therein.
I put the data in these,thanks
https://github.com/aromring/MAX30102_by_RF/compare/master...GOKUandVEGET:patch-1

GOKUandVEGET, 2020-04-11:
// RF, remove linear trend (baseline leveling)
  beta_ir = rf_linear_regression_beta(an_x, mean_X, sum_X2);
  beta_red = rf_linear_regression_beta(an_y, mean_X, sum_X2);
  for(k=0,x=-mean_X,ptr_x=an_x,ptr_y=an_y; k<n_ir_buffer_length; ++k,++x,++ptr_x,++ptr_y) {
    *ptr_x -= beta_ir*x;
    *ptr_y -= beta_red*x;
	  printf("%f ", *ptr_x);
  }

I printf the *ptr_x data

aromring, 2020-04-11:
OK, thank you for the raw data. In there, you write for example: "hr=71, but data show 142". Which number is correct, 71 or 142?

GOKUandVEGET, 2020-04-11:
> OK, thank you for the raw data. In there, you write for example: "hr=71, but data show 142". Which number is correct, 71 or 142?

The heart rate calculated in the algorithm is 71, but according to the waveform and the actual heart rate is 142

GOKUandVEGET, 2020-04-11:
> OK, thank you for the raw data. In there, you write for example: "hr=71, but data show 142". Which number is correct, 71 or 142?

My parameters，FS=50,FS60=3000

aromring, 2020-04-12:
Finally, in the 7th message you are sending useful information. I was about to close this issue, because 5 peaks in 4 seconds, as the plot of your data shows:
[image]
is very much consistent with heart rate = 71 bpm. 
But now you are telling me that you have changed the FS parameter, so it's a different story - with double sampling frequency, FS=50, the heart rate should be 142, indeed.
My first question is, therefore:
1) Exactly _which_ header file did you edit:  algorithm.h or algorithm_by_RF.h ? 
Next:
2) What other parameters did you change?
3) What settings did you use (i.e., status of all #define's in RD117_ARDUINO.ino
4) What is your hardware setup (MCU, connections, etc.)?

aromring, 2020-04-13:
5. You sent me 100 samples per measurement. It means that along with FS=50 you must have changed the ST parameter (sampling time in seconds) to 2. Correct?
6. Did you actually set MAX30102 sensor to a higher sample rate by writing to its REG_FIFO_CONFIG and/or REG_SPO2_CONFIG registers?

aromring, 2020-04-13:
7. After changing FS and (perhaps) ST, did you recalculate the sum_X2 parameter?

GOKUandVEGET, 2020-04-13:
> 1. After changing FS and (perhaps) ST, did you recalculate the sum_X2 parameter?
> 2. Did you change MAX_HR and/or MIN_HR parameters?

1. I edited algorithm by RF. H

2.#define ST 2

#define FS 50

3. I transplanted algorithm by RF to MUC without using other algorithms

4. There is no error in hardware connection. I use stm32f103c8t6

5. yes.

6. In max30102. C, I modified this parameter max30102 bus write (reg FIFO config, 0x2F); / / sample AVG = 2. Other parameters are the same as yours

7.50 * 2 = 100, your original is 25 * 4 = 100, so I don't need to modify the sum_X2 parameter

8. Min_HR = 40, I tried to change max_HR to 125, and I also tried to change it to 200, but it's still wrong under high heart rate

aromring, 2020-04-13:
Thank you for all the answers! That helped a lot. I have just changed one line of code in algorithm_by_RF.cpp that might fix this bug. Definitely, MAX_HR should sizably exceed the TYPICAL_HR, so for the latter = 120 set MAX_HR to at least 180. Try the new code and let me know if it fixes the problem.

GOKUandVEGET, 2020-04-13:
#define MAX_HR 200
#define MIN_HR 40   
#define TYPICAL_HR 120

I try to exercise for a little while to make sure my heart rate is greater than 100. Then I take a static measurement. At the beginning, the first and second data will occasionally show data greater than 100, but most of the time, the data displayed is still within 65.（I'm sure my heart rate is greater than 100）

aromring, 2020-04-13:
All right, set your code to save raw data for every measurement and send me the case you describe: factually high HR, but code reporting half of the number.

GOKUandVEGET, 2020-04-14:
This data completely uses your algorithm. 50Hz, 2S, typical heart rate 120, maximum heart rate 200, minimum heart rate 40. (the data I sent you before is filtered by me, and this data completely uses your algorithm, but the parameters are modified.)

Heart rate detected 69, actual 138
128.5422 | 160.4525 | 186.3627 | 214.273 | 238.1833 | 254.0936 | 282.0039 | 317.9142 | 354.8245 | 379.7348 | 339.6451 | -3.44464 | -546.534 | -674.624 | -654.714 | -558.803 | -433.893 | -253.983 | -59.0726 | 135.8377 | 253.748 | 318.6583 | 355.5685 | 392.4789 | 418.3892 | 437.2994 | 478.2097 | 153.12 | 185.0303 | 204.9406 | 235.8509 | 254.7612 | 43.67147 | -521.418 | -779.508 | -793.598 | -714.687 | -604.777 | -448.867 | -255.956 | -102.046 | 17.86409 | 83.77438 | 125.6847 | 161.595 | 169.5052 | 179.4155 | 203.3258 | 202.2361 | 211.1464 | 230.0567 | 253.967 | 282.8773 | 162.7876 | -351.302 | -696.392 | -732.482 | -715.571 | -633.661 | -510.751 | -352.84 | -189.93 | -35.0198 | 69.89049 | 140.8008 | 180.7111 | 206.6214 | 200.5316 | 217.4419 | 226.3522 | 220.2625 | 252.1728 | 295.0831 | 343.9934 | 326.9037 | -15.186 | -496.276 | -622.365 | -636.455 | -575.545 | -480.635 | -331.724 | -153.814 | 37.09631 | 173.0066 | 245.9169 | 287.8272 | 289.7375 | 302.6478 | 294.558 | 309.4684 | 320.3786 | 340.2889 | 371.1992 | 406.1095 | 430.0198 | 395.9301 | -8.15962 | -421.249 | -527.339

Heart rate detected 66, actual 132
-647.629 | -596.828 | -517.028 | -368.227 | -226.427 | -63.6262 | 72.17442 | 143.975 | 165.7756 | 183.5762 | 179.3768 | 171.1774 | 179.978 | 206.7785 | 232.5791 | 252.3797 | 280.1803 | 330.9809 | 366.7815 | 27.58206 | -377.617 | -469.817 | -495.016 | -444.216 | -366.415 | -240.614 | -96.8138 | 67.98677 | 201.7874 | 268.588 | 286.3886 | 281.1891 | 262.9897 | 267.7903 | 276.5909 | 279.3915 | 295.1921 | 331.9926 | 360.7932 | 398.5938 | 417.3944 | 177.195 | -242.004 | -415.204 | -462.403 | -428.603 | -379.802 | -286.001 | -191.201 | -50.4003 | 91.40029 | 190.2009 | 240.0015 | 228.8021 | 193.6026 | 157.4032 | 152.2038 | 173.0044 | 185.805 | 193.6056 | 234.4062 | 278.2068 | 336.0074 | 265.8079 | -90.3915 | -324.591 | -428.79 | -439.99 | -402.189 | -341.389 | -240.588 | -132.787 | -36.9868 | 41.81382 | 92.61441 | 97.415 | 77.21558 | 38.01617 | 23.81676 | 26.61736 | 57.41794 | 79.21852 | 117.0191 | 153.8197 | 209.6203 | 253.4209 | 116.2215 | -158.978 | -329.177 | -392.377 | -395.576 | -346.776 | -275.975 | -166.174 | -55.3738 | 55.42676 | 143.2273 | 168.0279 | 161.8285 | 123.6291

Heart rate detected 63, actual 126
35.41824 | 30.92289 | 49.42754 | 66.93219 | 80.43684 | 113.9415 | 158.4461 | 191.9508 | 234.4554 | 151.9601 | -128.535 | -307.031 | -397.526 | -409.021 | -368.517 | -301.012 | -206.507 | -110.003 | 2.501949 | 57.0066 | 71.51125 | 68.0159 | 38.52055 | 18.0252 | 21.52985 | 19.0345 | 56.53915 | 94.0438 | 114.5484 | 158.0531 | 197.5578 | 244.0624 | 171.567 | -79.9283 | -268.424 | -338.919 | -360.414 | -332.91 | -248.405 | -171.9 | -60.3957 | 45.10891 | 101.6136 | 123.1182 | 92.62286 | 68.12751 | 53.63216 | 56.13681 | 89.64146 | 134.1461 | 170.6508 | 218.1554 | 253.6601 | 325.1647 | 359.6694 | 369.174 | 143.6787 | -84.8167 | -225.312 | -260.807 | -254.303 | -185.798 | -113.293 | -15.7888 | 84.71587 | 150.2205 | 160.7252 | 150.2298 | 108.7345 | 82.23912 | 77.74377 | 67.24843 | 87.75308 | 126.2577 | 152.7624 | 168.267 | 199.7717 | 246.2763 | 273.781 | 179.2856 | -59.2097 | -247.705 | -324.2 | -353.696 | -318.191 | -278.686 | -206.182 | -123.677 | -63.1725 | -18.6679 | -19.1632 | -33.6586 | -47.1539 | -75.6493 | -58.1446 | -58.64 | -4.13532 | 28.36933 | 58.87399 | 65.37864

GOKUandVEGET, 2020-04-14:
[新建 Microsoft Excel 工作表 (2).xlsx](https://github.com/aromring/MAX30102_by_RF/files/4473092/Microsoft.Excel.2.xlsx)
data

aromring, 2020-04-14:
Thank you for the data. Unfortunately, I am very busy right now and debugging analysis will have to wait until weekend.

GOKUandVEGET, 2020-04-15:
Based on your latest documents, I have made the following changes :n_last_peak_interval = INIT_INTERVAL ;
My parameters：50hz 3s
Here's the data
ir data（Heart rate from 100 to 76, I'm not exercising, it's the same state. The difference is a little big,）
113115 113125 113147 113198 113226 113210 113182 113168 113150 113146 113141 113131 113159 113155 113183 113181 113208 113204 113249 113249 113267 113268 113285 113285 113242 113160 113063 112976 112921 112893 112885 112892 112917 112916 112947 112942 112941 112931 112904 112891 112892 112889 112901 112900 112942 112965 113002 113028 113028 113040 113063 113048 113057 113045 112987 112906 112835 112789 112758 112762 112758 112778 112800 112808 112815 112812 112798 112783 112786 112780 112779 112782 112807 112811 112825 112835 112849 112850 112863 112882 112886 112898 112900 112924 112935 112886 112805 112736 112671 112647 112638 112642 112668 112679 112698 112716 112715 112691 112692 112691 112675 112659 112681 112680 112704 112712 112734 112746 112754 112761 112771 112792 112800 112805 112819 112804 112739 112673 112620 112580 112578 112579 112594 112637 112650 112668 112656 112645 112646 112630 112630 112623 112641 112634 112649 112663 112678 112693 112714 112722 112722 112745 112749 112761 112781 112768 112709 112659 112594 112529 **************HR=100, HRvalid=1, SpO2=99, SPO2Valid=1
112497 112488 112493 112501 112527 112525 112545 112535 112529 112512 112497 112496 112499 112502 112512 112521 112540 112552 112566 112576 112595 112608 112595 112616 112625 112637 112604 112534 112436 112382 112338 112319 112318 112319 112334 112343 112357 112354 112341 112343 112318 112307 112296 112300 112304 112311 112325 112338 112349 112355 112379 112373 112390 112412 112409 112420 112430 112444 112430 112361 112273 112192 112139 112100 112093 112093 112098 112114 112140 112144 112146 112130 112111 112098 112068 112064 112069 112093 112108 112107 112126 112134 112135 112155 112176 112184 112202 112222 112245 112258 112237 112174 112077 112002 111952 111936 111922 111955 111956 111974 112003 112020 112010 111983 111976 111967 111964 111967 111990 112005 112017 112027 112057 112076 112103 112126 112143 112159 112185 112183 112211 112231 112223 112156 112077 112008 111959 111926 111929 111938 111960 111990 112005 112024 112023 112022 112010 111998 111993 111995 112015 112036 112059 112081 112108 112135 112161 112181 112194 112216 **************HR=93, HRvalid=1, SpO2=99, SPO2Valid=1
112245 112248 112285 112310 112316 112322 112260 112190 112103 112062 112037 112032 112024 112039 112056 112081 112096 112092 112071 112077 112057 112060 112062 112080 112104 112124 112154 112184 112198 112224 112238 112260 112283 112305 112321 112347 112377 112404 112407 112336 112276 112200 112162 112139 112138 112154 112169 112180 112217 112228 112221 112218 112212 112198 112190 112207 112202 112233 112265 112288 112312 112343 112361 112374 112400 112415 112447 112461 112478 112505 112508 112475 112409 112340 112275 112246 112229 112239 112248 112255 112279 112316 112317 112322 112311 112298 112292 112301 112311 112310 112350 112367 112386 112433 112452 112463 112479 112493 112502 112527 112538 112551 112565 112541 112469 112378 112323 112273 112259 112258 112262 112278 112289 112304 112309 112320 112292 112280 112279 112271 112293 112304 112298 112320 112339 112369 112386 112418 112424 112441 112472 112476 112502 112524 112538 112546 112493 112411 112344 112282 112267 112255 112261 112281 112301 112313 112333 112326 112310 112305 **************HR=90, HRvalid=1, SpO2=99, SPO2Valid=1
112292 112269 112265 112279 112282 112309 112305 112353 112351 112377 112374 112403 112416 112429 112445 112456 112418 112335 112246 112195 112156 112152 112170 112178 112199 112213 112239 112238 112238 112230 112231 112218 112227 112245 112251 112260 112275 112311 112334 112350 112379 112382 112411 112427 112434 112466 112481 112505 112495 112426 112349 112263 112202 112164 112159 112177 112192 112200 112227 112239 112225 112206 112183 112158 112153 112154 112156 112185 112184 112218 112230 112257 112276 112299 112320 112325 112350 112372 112366 112319 112243 112155 112094 112072 112050 112057 112062 112081 112117 112136 112129 112121 112104 112095 112110 112119 112126 112144 112166 112173 112201 112221 112234 112258 112272 112302 112309 112338 112363 112377 112354 112283 112175 112094 112030 111985 111993 111997 112013 112031 112046 112048 112043 112038 112019 112004 112000 111998 112016 112032 112047 112068 112088 112108 112111 112139 112148 112177 112195 112216 112195 112138 112031 111937 111869 111829 111817 111819 111836 111870 **************HR=96, HRvalid=1, SpO2=99, SPO2Valid=1
111894 111902 111900 111895 111876 111877 111866 111888 111892 111918 111935 111952 111976 112014 112028 112061 112087 112109 112123 112141 112164 112156 112191 112210 112154 112059 111959 111863 111808 111784 111801 111818 111840 111882 111887 111890 111881 111865 111853 111831 111846 111849 111865 111894 111912 111931 111962 111985 112018 112030 112046 112073 112103 112112 112085 111993 111907 111821 111765 111750 111754 111772 111788 111820 111853 111861 111869 111867 111849 111855 111859 111864 111875 111900 111918 111955 111974 112008 112054 112062 112087 112112 112124 112155 112178 112185 112196 112214 112245 112271 112300 112262 112163 112041 111941 111865 111823 111828 111835 111852 111865 111904 111920 111918 111895 111878 111872 111869 111881 111899 111922 111935 111959 111986 112015 112037 112041 112056 112085 1 [... cut ...] 112158 112174 112177 112205 112238 112275 112294 112291 112213 112104 111987 111904 111846 111825 111851 111843 111872 111908 111925 111930 111923 111908 111904 111900 111907 111918 111940 111977 112008 112039 112069 112089 112117 112151 112173 112195 112214 112230 112242 112271 112301 112322 112340 112372 112392 112400 112337 112216 112087 111990 111953 111926 111936 111938 111961 111994 112010 112018 112010 112003 111981 111969 111982 111995 **************HR=76, HRvalid=1, SpO2=99, SPO2Valid=1

GOKUandVEGET, 2020-04-15:
> Thank you for the data. Unfortunately, I am very busy right now and debugging analysis will have to wait until weekend.

Thank you very much for your reply and look forward to your reply at the weekend

GOKUandVEGET, 2020-04-15:
The problem seems to be that the heart rate shown in the previous groups of data is not accurate, and the latter is basically stable

aromring, 2020-04-19:
As I understand, after you implemented my latest fix (n_last_peak_interval = INIT_INTERVAL) the heart rate algorithm consistently starts from your TYPICAL_HR = 120 and reports high heart rates correctly. I have verified it with all data you sent me in the Excel spreadsheet. 
As for the second batch of data, I have analyzed all series and found no error in calculations. For example, the second series where HR=93 looks like this when plotted:
[image]
The top graph is the raw signal, the bottom represents the same signal after normalization. Convince yourself that the distance between peaks is 57 - 25 = 32 samples, yes? With your sampling rate of 50 Hz the peaks are 32/50 = 0.64 seconds apart. It means 1.5625 heartbeats per second, or 93.75 bpm, correct? Again, all this is being concluded from the _raw_ signal, before it even enters my code.
Let's see if my algorithm agrees with this result. Below is the autocorrelation graph for the same signal:
[image]
The initial point is TYPICAL_HR = 120 and this corresponds to the INIT_INTERVAL = 25. Starting from 25, the HR algorithm climbs uphill until it reaches the peak at 32. Which, as you know, is converted to HR = 93.75 bpm; as expected. Thus, the code simply reports whatever the raw signal contains on input, no more and no less.
In conclusion, the high HR variability originates in the MAX30102 sensor itself. Only Maxim, Inc. can adequately address these concerns.

aromring, 2020-04-23:
I have not heard from the original posters in 3 days, this I consider this bug as fixed.

closed 2020-04-23 by aromring