aromring/MAX30102_by_RF #18: New code bug

GOKUandVEGET, 2020-06-06:
When I used your new code, I found that the heart rate still showed half of the actual result in the occasional test. This kind of probability is very low, but it still exists, so I can't record the data to analyze, so I wonder if it's because the setting of relative autocorrelation 0.5 is too large? According to your algorithm, as long as the initial detected heart rate is half of the actual value, the subsequent detected heart rate should be only half of the actual value.

GOKUandVEGET, 2020-06-06:
At that time, my actual heart rate was more than 80, and the detection value was more than 40. When I released my hand and tested again, the correct heart rate was more than 80

aromring, 2020-06-06:
Yes, I agree that even occasional cases of this old bug resurfacing are the cause for concern. I will be very happy to debug the issue and fix it (after I return from my travels), but as usual I need raw data that lead to erroneous HR values. You can uncomment the #define SAVE_RAW_DATA in main code to display all the raw data along with results in the Serial monitor. Then you can fish out sensor data resulting in wrong HR and send these to me.

aromring, 2020-08-24:
No response since early June probably indicates a successful resolution.

closed 2020-08-24 by aromring