aromring/MAX30102_by_RF #21: new code bug

GOKUandVEGET, 2020-08-31:
When I used your new code, I found that the heart rate still showed half of the actual result in the occasional test. This kind of probability is very low, but it still exists, so I can't record the data to analyze, so I wonder if it's because the setting of relative autocorrelation 0.5 is too large? According to your algorithm, as long as the initial detected heart rate is half of the actual value, the subsequent detected heart rate should be only half of the actual value.

50hz，3s

There are more than 150 data here because I also copied the data that was not passed through the algorithm

my hr=120,bur show 60

raw ir data
110491 110550 110537 110512 110505 110535 110597 110704 110778 110778 110811 110883 110936 110932 110978 111074 111153 111138 110949 110722 110630 110649 110766 110838 110882 110972 110997 110939 110887 110832 110814 110851 110932 111034 111078 111086 111162 111226 111225 111233 111322 111379 111218 110965 110840 110881 110926 110956 110976 111018 111064 111106 111148 111183 111247 111305 111374 111405 111456 111522 111501 111484 111508 111515 111527 111559 111410 111145 111017 111052 111169 111275 111315 111361 111347 111289 111258 111302 111320 111351 111432 111496 111556 111586 111626 111620 111616 111667 111712 111659 111444 111250 111179 111264 111359 111387 111454 111531 111509 111474 111491 111499 111535 111584 111619 111658 111703 111750 111782 111793 111812 111863 111913 111936 111892 111730 111479 111350 111348 111396 111545 111665 111684 111666 111657 111638 111626 111675 111715 111750 111810 111869 111895 111917 111930 111916 111922 111971 112003 111844 111621 111532 111549 111590 111680 111795 111864 111851 111802 111782 111792 111832 111874 111912 111987 112035 112022 112013 112099 112153 112146 112133 112252 112451 112317 111959 111680 111715 111803 111824 111875 111979 111999 111953 111888 111906 111953 111988 112019 112068 112134 112149 112136 112123 112159 112174 112196 112246 112207 111938 111659 111602 111705 111831 111931 112041 112074 112059 112029 112021 112036 112116 112215 112292 112329 112377 112454 112500 112488 112493 112510 112508 112526 112575 112613 112674 112570 112257 112051 112061 112138 112200 112304 112426 112440 112378 hr：60 spo2：98

112338 112302 112313 112381 112478 112549 112592 112630 112669 112693 112743 112735 112727 112646 112417 112221 112202 112300 112420 112494 112563 112592 112567 112531 112507 112531 112585 112657 112700 112737 112791 112825 112804 112810 112793 112784 112797 112861 112861 112605 112373 112330 112378 112459 112538 112623 112761 112777 112715 112683 112673 112698 112751 112885 112993 113016 112989 112982 112979 113018 113092 113146 113139 112861 112546 112422 112451 112546 112655 112755 112837 112846 112804 112755 112740 112762 112816 112936 113016 113054 113068 113085 113135 113183 113206 113092 112811 112618 112568 112635 112765 112868 112967 113027 113034 112981 112928 112932 112957 113029 113126 113217 113227 113223 113261 113344 113370 113331 113072 112775 112655 112665 112775 112887 113003 113081 113099 113072 112997 112982 113074 113163 113232 113251 113294 113346 113386 113412 113434 113473 113341 113048 112898 112895 112956 113031 113141 113251 113280 113214 113145 113156 113194 113253 113368 113469 113465 113467 113505 113549   hr：65  spo2：99

aromring, 2020-09-13:
Make the following changes to the code algorithm_by_RF.cpp:

1) Comment out line 98:

`//    if(LOWEST_PERIOD==n_last_peak_interval)`

2) Insert a new line 99:

`n_last_peak_interval=LOWEST_PERIOD`

This way the `rf_initialize_periodicity_search()` will run every time at the cost of slower performance. But at least the code will be immune to sudden HR changes.

GOKUandVEGET, 2020-09-13:
Thank you for your answer, but this does not solve the potential bug. According to the data I gave, in some special cases, after your algorithm, it will point to the second local maximum value of autocorrelation, so that the detection heart rate is half of the actual value. The subsequent calculation depends on the initialization range, and the detection heart rate is half of the actual value. Your solution is just to look for the initial heart rate interval every time, instead of solving the case where the algorithm points to the second local maximum of autocorrelation in some special cases.

aromring, 2020-09-13:
Have you even tried my suggestion before dismissing it upfront? Initializing the algorithm to the LOWEST_PERIOD means initializing it to the _highest_ heart rate every time. Thus, detecting the first peak is also guaranteed every time.

Besides, this issue is a duplicate of existing issue #18

closed 2020-09-13 by aromring