aromring/MAX30102_by_RF #22: MAX30102 FIFO returns IR data first

alireza9900, 2020-09-24:
Right now I’m working on MAX30102 with the MH-ET LIVE breakout board for it and I’ve noticed that when you drive the module in the heart rate mode (i.e. 0x02 mode configuration) the IR led becomes active. The datasheet states that in this mode only the red led should be working. Furthermore the data read from the FIFO is ir data first and then the red data. Curiously, in the MAX30100 this exact behaviour is expected. (I’m refering to its datasheet and haven’t confirmed this physically.) I’m getting part id 0x15 from the module I’m working with so it is indeed MAX30102 (or 101 or 105, but these are all the same in this regard) and not a MAX30101. I possibly can’t think of anything that could cause this behaviour except that the info on the datasheet for MAX30102 is wrong and surprisingly everybody is using this module wrong, including the guys at Maxim Integrated! Obvisouly reading raw IR and Red data instead of each other results in wrong calculation of SpO2 levels.

So have anyone noticed this behaviour in MAX30102?

P.S.: There is a chance that I might have a faulty revision of the module. My part returns Revision ID: 0x03.

aromring, 2020-09-24:
Some cheap clones have their channels swapped, see this issue: https://github.com/aromring/MAX30102_by_RF/issues/13#
Other than that, your post belongs to the MAXIM hardware forum.

closed 2020-09-24 by aromring

alireza9900, 2020-09-25:
@aromring Thank you for letting me know of the related issue.