aromring/MAX30102_by_RF #23: Accurate HR and inaccurate spo2 with inaccurate low spo2 test sample

MirzaSamnani, 2020-10-10:
Hey there!. Firstly thanks for a lovely algo! Secondly sorry if I didn't follow the troubleshooting guide as that is not the case here. Everything working like a charm, Also I know this does not have any liability but we could just make it more accurate/reliable.

Using Arduino Mega, Chinese clone MH-ET 30102
for normal individuals the spo2 and heartrate are accurate. 

However, I tested the values with a person whose spo2 was 89-92, and the values on the algo shot 97-98. 
the other Maxim algo is absolute garbage for both MAX30100 30102 where the values hit 98-100 even when the spo2 was around 92. One possibility is that people have not tested these algos/ devices on patients with low saturation. However, I feel when my O2 was 99.6 ish and it did drop to 97 ish for a patient with low O2 around 90 ish, you/we could tweak around something to get more accuracy in that region. I am happy to go back to the COVID area and test on another patient if you have any changes. 

My observation so far(for RF algo)-

Healthy patient O2 reported - 99.8 ish Actual O2 from a medical monitor- 98 ish
Low O2 patient O2 reported - 97 ish Actual O2 from a medical monitor- 91 ish

Any thoughts?

aromring, 2020-10-12:
Well... First, it seems to me you are using hypoxemic COVID-19 patients to test your homemade SpO2 sensor that utilizes a cheap clone MH-ET 30102 and free algorithm posted on Instructables.com... Is this correct? I will leave ethical implications of this activity to you and your consciousness. 
Second, no sensor measures SpO2 directly - the oxygen saturation must be calculated from the relative RMS intensities of both signals (the "Z" value). How? Through an empirical calibration. I used the empirical SpO2(Z) function directly from [MAXREFDES117#](https://www.maximintegrated.com/en/design/reference-design-center/system-board/6300.html/tb_tab1) reference design by MAXIM, Inc. You can see it all in MAXIM's [algorithm.h](https://github.com/aromring/MAX30102_by_RF/blob/master/algorithm.h), but clicking on the previous link yields further information:
"SpO2 calculation is based on the equation shown below. However, determining the constants (C1, C2, and C3) requires a comprehensive clinical study of pulse pulse-oximetry data from a statistically significant population set using this hardware. Such a clinical study is beyond scope of this design. Therefore, the calculated SpO2 value may have an error."
How did MAXIM determine their C1, C2, and C3 constants? They don't elaborate, but I can imagine them using a few of their _healthy_ employees as volunteers. Now, I must stress that all of this applies to the _original_ MAX30102 sensor, _not_ the cheap clone.
So. If a large corporation like MAXIM can't afford a statistically significant clinical study, what makes you think that an individual amateur like myself can? What do you mean by me "tweaking around something to get more accuracy"? I am not a psychic with an omniscient crystal ball capable of picking proper calibration out of thin air! I hope you were just joking.
If you have means to conduct such study, then you have found answer to your questions. Recalibration would be exceptionally easy: you would just put your new numbers in the line 121 of [algorithm_by_RF.cpp](https://github.com/aromring/MAX30102_by_RF/blob/master/algorithm_by_RF.cpp).

closed 2020-10-12 by aromring

MirzaSamnani, 2020-10-12:
Firstly, as a matter of goodwill while testing a couple of medical grade probes such as AFE4490 etc in my device. I found this issue and thought to give you a heads up since you worked so hard on this. Secondly, I utilized Protocentral MAX30102(which uses a genuine MAX30102) and MH ET LIVE which I found was a clone after seeing its IR values are swapped. My custom extension of the MAX algo gave fairly accurate readings for such patients but not for healthy individuals and with more fluctuations and hence reached out to you.

Anyway thanks.

aromring, 2020-10-13:
Hi Mirza,
Essentially, all the issues raised on this forum, as well as questions asked on the related Instructables page, come from amateur hobbyists, like myself. Hence, I honestly thought you were just a hobbyist testing his homemade device(s) on patients.
But a web search revealed that you actually do it professionally, so I apologize for a harsh tone of my first response.
The bottom line: you need to collect a lot of data points to recalibrate the algorithm.

MirzaSamnani, 2020-10-14:
Hey! Indeed yeah, after making a few bots for the Covid hospitals, I had thought of developing low-cost spo2 trackers which could potentially give a response to those on O2 supply much in advance then in practice. AFE turned out to be higher on the budget. 

And I was completely surprised to see the values coming from MAXIM algorithm both on MAX30100 and 30102 where the spo2 was 88 and it reported 99/100% and my algo failed on healthy patients. Yours was pretty close on the change of values. The main problem was that you have to wear a PPE and have roughly 45mins to an hour to do what ever you can to tweak/ collect data. Within 15 minutes the glasses start fogging out.

No worries though I got a really good starting point. It will surely help recalibrate the algo.

Thanks again and cheers mate!