aromring/MAX30102_by_RF #25: I cant found more detail sensor settings, problems with get values

sjsivo, 2021-02-26:
BPM and SPO2 Values are not valid for my sensor MAX30102 and ESP32.
Used more types of algoritm and result is not valid BPM and SPO2 values.
How can i correct?

[
[image]
[image]
[image]
[image]

](url)

aromring, 2021-02-27:
You can start by explaining what you exactly mean by "BPM and SPO2 Values are not valid" and what the blue and red curves represent. Thank you for the graphs, by the way.

sjsivo, 2021-02-27:
I reading the IR and Red diode values from Max30102 sensor on ESP32 chip.
Red and blue are values from this 2 lights from Max chip showed in Serial Plotter.
I mean: algoritm can not recognize that some heartbeat come and can not calculate SPO2. Base on Current and lightning from diods recognize sometimes that BPM can calculate sometimes, but much more times are values not valid.
With this i mean is, that on display or in Serial communication ,algoritm showed values "-999" and notvalid variable are set. 
Can i change default settings for el. current to diods and retest with more different settings?
I cant read good values from original Maxim algoritm too.
How cann i correct this situation?

sjsivo, 2021-02-27:
I have change more times ir and red current values in max30102.cpp
. Still i have no luck to measure SPO2. BPM is sometimes shown correctly now.

[image]
[image]

sjsivo, 2021-02-27:
In algorithm_by_RF.cpp was checked xy_ratio too...

Last measured values here:

[valuesMeasured.txt](https://github.com/aromring/MAX30102_by_RF/files/6054237/valuesMeasured.txt)

aromring, 2021-02-28:
Ah, yes. Thank you for providing the xy_ratio - this saves days of e-mail exchanges. High values of this parameter clearly indicate a well known problem: you are using a cheap clone of MAX30102 in which the IR and RED channels are swapped. See issue #13 for more details:
https://github.com/aromring/MAX30102_by_RF/issues/13#issue-601473302

closed 2021-02-28 by aromring

sjsivo, 2021-02-28:
Now is good.
Its measure correctly with original values too.
I changed:
 for(i=0;i<BUFFER_SIZE;i++)
  {
    while(digitalRead(oxiInt)==1);  //wait until the interrupt pin asserts
    maxim_max30102_read_fifo( (aun_ir_buffer+i),(aun_red_buffer+i));  //read from MAX30102 FIFO //swap values for cheap modules!!!

in your sample code and works correctly.
Thank for your helpful support.