aromring/MAX30102_by_RF #26: HR always -999 or invalid while wearing sensor on wrist.

AAJAY5, 2021-03-31:
I have checked/verified listed below thing before raising issue.
- Sensor is original not swapped (Tested).
- When I put sensor on finger/thumb it works fine and give accurate readings. I also compared with amaze fit and looks fine. But, when i put sensor on my wrist using my watch case assembly it's always read invalid HR.
- I have attached reading sample as .xlsx file.
- Can you please check and verify everything is expected or not

Thanks,
Have a good day.
[reading.xlsx](https://github.com/aromring/MAX30102_by_RF/files/6236710/reading.xlsx)

aromring, 2021-04-01:
Your spreadsheet does not contain plots of raw signals. Please fix this issue and resubmit. 
If you need context, display the README page at https://github.com/aromring/MAX30102_by_RF/blob/master/README.md  and scroll down to "HOW TO REPORT BUGS".

AAJAY5, 2021-04-02:
I have updated sheet with chart. What do you think about max30102 work on wrist or not?

AAJAY5, 2021-04-02:
> Your spreadsheet does not contain plots of raw signals. Please fix this issue and resubmit.
> If you need context, display the README page at https://github.com/aromring/MAX30102_by_RF/blob/master/README.md and scroll down to "HOW TO REPORT BUGS".

I have updated sheet for sensor on finger as well as sensor on wrist with charts. Can you please look at that and let me know is everything is okay or not?

aromring, 2021-04-02:
Excellent. Now you know why I want people to draw charts before submitting bug reports: I wanted you to see for yourself that the quality of signal is dismal when sensor is placed on wrist (IR in particular). Don't expect any algorithm to read good HR from poor signals. I don't think the sensor was designed for wrist - but to be sure contact MAXIM.

closed 2021-04-02 by aromring

AAJAY5, 2021-04-05:
I have updated sketch with web socket and graph plot for every 100 readings. So, graph automatically updates @4s. So, after 1-2 minutes sensor gives reading/HR value on wrist.
But on even finger movement reading get unstable and takes 1-2 minutes to get stable.

[image]

aromring, 2021-04-07:
Awesome, thank you for sharing! Indeed, motion artifacts always interfere, even when the sensor is on a fingertip. Wrists are much worse because of all the moving tendons under the skin. I wonder how Apple Watch and other "fitness sensors" deal with these problems?

AAJAY5, 2021-04-07:
Motion artifacts affect play important role in PPG. I have added accelerometer to hardware. I am just informing here because i wanted to contribute. Currently i am just co-relating things with accel data. I have attached screen shot for the same. If you have any info about this please let me know/share.
[image]

aromring, 2021-04-09:
Mighty interesting! Thank you for sharing. I did think about adding an accelerometer to mitigate the effect of motion artifacts, but I've never had sufficient motivation to do it. Besides, it's not that obvious how to do it. Accelerometer measures the movement of the whole arm/body, while motion artifacts are mainly caused by wrist tendons that move fingers/hand. The two may or may not be correlated...

AAJAY5, 2021-05-14:
I have purchased MAX30105 sensor. Is that algorithm is valid for the same green led instead of red one? Any changes for green led?

aromring, 2021-05-15:
I have no idea. I've never used MAX30105.

AAJAY5, 2021-05-17:
> I have no idea. I've never used MAX30105.

Okay. Thanks...

yuelsun, 2025-03-18:
Hello AAJAY5, I would like to ask if the effect of heart rate inaccuracy caused by motion artifacts has been resolved?