aromring/MAX30102_by_RF #28: RED/IR sensor value looks strange

integer-bin, 2021-05-31:
Good morning, sir. Currently, I am implementing your code using Atmega4809. However, my data value is very different from the data value you uploaded on the homepage. I saw the issue posted by the previous people, but no one seems to be in the same situation as me. I test whether the RED LED and IR LED are reversed, but my device was normal. Below is my data value that I received through UART communication. What could have gone wrong? I'll be waiting for the reply. Thank you, teacher.

0: red = 232060, ir = 242086
1: red = 231443, ir = 241347
2: red = 232013, ir = 242024
3: red = 232013, ir = 242024
4: red = 231429, ir = 241330
5: red = 231978, ir = 241986
6: red = 231978, ir = 241986
7: red = 231978, ir = 241986
8: red = 231412, ir = 241316
9: red = 231997, ir = 241989
10: red = 231997, ir = 241989
11: red = 231376, ir = 241246
12: red = 232078, ir = 242055
13: red = 232078, ir = 242055
14: red = 232078, ir = 242055
15: red = 232078, ir = 242055
16: red = 232062, ir = 242047
17: red = 232062, ir = 242047
18: red = 231245, ir = 241130
19: red = 232080, ir = 242092
20: red = 232080, ir = 242092
21: red = 231257, ir = 241134
22: red = 232012, ir = 242014
23: red = 232012, ir = 242014
24: red = 232012, ir = 242014
25: red = 231276, ir = 241173
26: red = 232006, ir = 241941
27: red = 232006, ir = 241941
28: red = 231301, ir = 241180
29: red = 231932, ir = 241870
30: red = 231932, ir = 241870
31: red = 231392, ir = 241290
32: red = 231783, ir = 241756
33: red = 231783, ir = 241756
34: red = 231783, ir = 241756
35: red = 231373, ir = 241270
36: red = 231733, ir = 241553
37: red = 231733, ir = 241553
38: red = 231312, ir = 241161
39: red = 231518, ir = 241353
40: red = 231518, ir = 241353
41: red = 231518, ir = 241353
42: red = 231308, ir = 241198
43: red = 231322, ir = 241179
44: red = 231322, ir = 241179
45: red = 231304, ir = 241203
46: red = 231308, ir = 241085
47: red = 231308, ir = 241085
48: red = 231360, ir = 241197
49: red = 231200, ir = 240970
50: red = 231200, ir = 240970
51: red = 231200, ir = 240970
52: red = 231330, ir = 241248
53: red = 231106, ir = 240902
54: red = 231106, ir = 240902
55: red = 231300, ir = 241211
56: red = 231217, ir = 241027
57: red = 231217, ir = 241027
58: red = 231217, ir = 241027
59: red = 231305, ir = 241219
60: red = 231284, ir = 241110
61: red = 231284, ir = 241110
62: red = 231275, ir = 241180
63: red = 231313, ir = 241169
64: red = 231313, ir = 241169
65: red = 231285, ir = 241166
66: red = 231323, ir = 241173
67: red = 231323, ir = 241173
68: red = 231323, ir = 241173
69: red = 231278, ir = 241215
70: red = 231287, ir = 241147
71: red = 231287, ir = 241147
72: red = 231259, ir = 241189
73: red = 231352, ir = 241172
74: red = 231352, ir = 241172
75: red = 231352, ir = 241172
76: red = 231331, ir = 241227
77: red = 231331, ir = 241227
78: red = 231331, ir = 241227
79: red = 231195, ir = 241122
80: red = 231356, ir = 241236
81: red = 231356, ir = 241236
82: red = 231180, ir = 241124
83: red = 231427, ir = 241244
84: red = 231427, ir = 241244
85: red = 231427, ir = 241244
86: red = 231194, ir = 241102
87: red = 231415, ir = 241336
88: red = 231415, ir = 241336
89: red = 231141, ir = 240973
90: red = 231421, ir = 241283
91: red = 231421, ir = 241283
92: red = 231063, ir = 240899
93: red = 231422, ir = 241313
94: red = 231422, ir = 241313
95: red = 231422, ir = 241313
96: red = 231068, ir = 240892
97: red = 231407, ir = 241317
98: red = 231407, ir = 241317
99: red = 231026, ir = 240774

aromring, 2021-06-01:
Open the README page for this repository:
https://github.com/aromring/MAX30102_by_RF
and scroll down to the "HOW TO REPORT BUGS" section.

aromring, 2021-10-05:
No response since May 31, 2021. Closing the issue.

closed 2021-10-05 by aromring

shashankholla, 2023-04-19:
Hi, were you able to find a fix for this? I am seeing a similar behaviour where data points are repeated