aromring/MAX30102_by_RF #32: Need to change 50 Sampling Frequency / 1 Sampling Time

madhavanatmel, 2022-04-12:
Hi, i  need to change the sampling Frequency to 50  / 1 Sampling Time .

#define ST 4      // Sampling time in s. WARNING: if you change ST, then you MUST recalcuate the sum_X2 parameter below!
#define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
// Sum of squares of ST*FS numbers from -mean_X (see below) to +mean_X incremented be one. For example, given ST=4 and FS=25,
// the sum consists of 100 terms: (-49.5)^2 + (-48.5)^2 + (-47.5)^2 + ... + (47.5)^2 + (48.5)^2 + (49.5)^2
// The sum is symmetrc, so you can evaluate it by multiplying its positive half by 2. It is precalcuated here for enhanced 
// performance. 

Already you mentioned the procedure to change bu t i could able to work out , its hanging if touch  rf algorithm .

could you please help me to share code  for 50 sampling frequency  in 1 second Sampling time.

aromring, 2022-04-14:
This forum is for reporting bugs in the code, thus in principle I should close this issue as not a bug. However, I am willing to give you the benefit of the doubt and a chance to state your problem in clear and understandable English. As written right now, I can barely understand what the problem is. 
Here is my guess: 
If you changed FS to 50 and retained ST=4, then you must recalculate sum_X2 to be the following:
(-99.5)^2 + (-98.5)^2 + (-97.5)^2 + ... + 97.5^2 + 98.5^2+ 99.5^2

madhavanatmel, 2022-04-15:
hi,
what the problem is. : HR not accurate .
From the (25 - samples X 4 - seconds ) algorithm will be enough to make good result but when we increase more sampling data possible to get high accurate result . i will apply above mentioned calculation method and i will come back with result comparison .

sergiocntr, 2022-04-15:
The HR is accurate as the hardware.... Glad to know which kind of board you use and if you own a Maxim probe or a clone. The lib works well for normal/amateur purposes... if you want to improve, better to buy a specific instrument I suggest , not trying to squeeze from lib what don't comes from HW.

madhavanatmel, 2022-04-16:
Hi, Interesting information about HW side .  i have purchased the sensor via online shopping .
[i](https://www.amazon.in/xcluma-MAX30102-Upgraded-Ultra-Low-Compatible/dp/B07TZCNG2G) 

i try buy spark fun board

aromring, 2022-04-16:
Good point, @sergiocntr. Thank you. 
All right, @madhavanatmel, if you suspect it's a bug in the code then you know the drill: I need all the information, including _plots_, mentioned in HOW TO REPORT BUGS section of https://github.com/aromring/MAX30102_by_RF/blob/master/README.md

aromring, 2022-06-27:
No response since April 16. Issue closed.

closed 2022-06-27 by aromring