aromring/MAX30102_by_RF #35: Error while compiling

sivaareddi, 2022-07-20:
i am getting this error when i compile 
```/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected identifier before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:34:7: note: in expansion of macro 'FS'
   34 | class FS;
      |       ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected unqualified-id before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:34:7: note: in expansion of macro 'FS'
   34 | class FS;
      |       ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected identifier before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:41: note: in expansion of macro 'FS'
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |                                         ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected ',' or '...' before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:41: note: in expansion of macro 'FS'
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |                                         ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected unqualified-id before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:127:5: note: in expansion of macro 'FS'
  127 |     FS                  *_baseFS;
      |     ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: default argument missing for parameter 2 of 'fs::File::File(fs::FileImplPtr, int)'
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:41: note: in expansion of macro 'FS'
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |                                         ^~
In file included from /home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:26,
                 from /home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h:30,
                 from /home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClientSecure.h:23,
                 from /home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiServerSecure.h:20,
                 from /home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/ESP8266WiFi.h:41,
                 from /home/csr/Documents/smart watch/max30102_test/max30102_test.ino:7:
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:22: note: ...following parameter 1 which has a default argument
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |          ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h: In constructor 'fs::File::File(fs::FileImplPtr, int)':
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:91: error: class 'fs::File' does not have any field named '_baseFS'
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |                                                                                           ^~~~~~~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:55:99: error: 'baseFS' was not declared in this scope
   55 |     File(FileImplPtr p = FileImplPtr(), FS *baseFS = nullptr) : _p(p), _fakeDir(nullptr), _baseFS(baseFS) { }
      |                                                                                                   ^~~~~~
In file included from /home/csr/Documents/smart watch/max30102_test/max30102_test.ino:4:
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h: At global scope:
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected identifier before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:132:41: note: in expansion of macro 'FS'
  132 |     Dir(DirImplPtr impl = DirImplPtr(), FS *baseFS = nullptr): _impl(impl), _baseFS(baseFS) { }
      |                                         ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected ',' or '...' before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:132:41: note: in expansion of macro 'FS'
  132 |     Dir(DirImplPtr impl = DirImplPtr(), FS *baseFS = nullptr): _impl(impl), _baseFS(baseFS) { }
      |                                         ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected unqualified-id before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
[... 63 lines cut ...]
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/FS.h:275:12: note: in expansion of macro 'FS'
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |            ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected identifier before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:48:27: note: in expansion of macro 'FS'
   48 |     int initCertStore(fs::FS &fs, const char *indexFileName, const char *dataFileName);
      |                           ^~
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected ',' or '...' before numeric constant
[... cut ...]
/home/csr/snap/arduino/current/Arduino/libraries/MAX30102_by_RF-master/algorithm_by_RF.h:45:12: error: expected unqualified-id before numeric constant
   45 | #define FS 25     // Sampling frequency in Hz. WARNING: if you change FS, then you MUST recalcuate the sum_X2 parameter below!
      |            ^~
/home/csr/snap/arduino/70/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/CertStoreBearSSL.h:54:9: note: in expansion of macro 'FS'
   54 |     fs::FS *_fs = nullptr;
      |         ^~
exit status 1
```

sivaareddi, 2022-07-20:
Arduino.h not found error then i removed arduino.h from all source code then i got this error can you please help?

sergiocntr, 2022-07-21:
This is not related to this lib.

By the way ,arduino.h has all the references needed for to compile the sketch,so re-add it.
Then
First choice , Arduino.h and arduino.h are not the same under Linux enviroment.
Second , try to build Blink sample ,then step by step add others things.

sivaareddi, 2022-07-23:
Thanks! changing all arduino.h to Arduino.h worked for me

closed 2022-07-23 by sivaareddi