aromring/MAX30102_by_RF #36: Float buffer allocation uses unnecessary memory

0xMihir, 2022-08-09:
The first part of the [`rf_heart_rate_and_oxygen_saturation`](https://github.com/aromring/MAX30102_by_RF/blob/master/algorithm_by_RF.cpp#L37) code allocates two float arrays to perform calculations on. The default buffer size of 100 results in 800 bytes (2 arrays * 4 bytes * 100 elements) of memory usage, making many libraries that require some dynamic memory impossible to use in conjunction with this one. 

I know that it's the best convention to leave parameters unmodified, but when working on underpowered microcontrollers like the Arduino Uno, performing all calculations on a single float buffer makes this library use a lot less memory. I have already modified the code to work this way, but I wanted to hear your thoughts before opening a PR.

aromring, 2022-08-11:
I am not exactly sure what you are referring to, so I am guessing it's about an_x and an_y arrays, correct? Indeed, some calculations can be performed separately for IR and RED channels using one array, but what about the rf_Pcorrelation() call? It needs both at the same time.

0xMihir, 2022-08-21:
Rather than creating the an_x and an_y arrays, I just modified the code to modify the arrays in the parameter. I realized that saying single float buffer was unclear.

aromring, 2022-08-27:
Thank you for good contribution, but I cannot accept the pull request as is because it seems incomplete - it does not contain matching changes to RD117_ARDUINO.ino. It still contains: 
uint32_t aun_ir_buffer[BUFFER_SIZE]; 
which is passed to your version of rf_heart_rate_and_oxygen_saturation() as float *. Perhaps there is an implicit conversion from uint32_t* but I would not bet that all compilers do it.

0xMihir, 2022-09-07:
Yep, I'll change it to `float`.

closed 2022-09-09 by aromring

aromring, 2022-09-09:
Hi 0xMihir,
I had to revert your pull request, sorry, since it would break the code in this repository. In fact, other functions in other parts of the code still use uint32_t incompatible with float. For example, maxim_max30102_read_fifo() in max30102.cpp or the original MAXIM algorithm in algorithm.cpp.

PR #37 (2022-09-09): perf: Only use Red & IR arrays from function param
This will close #36. Before, the `rf_heart_rate_and_oxygen_saturation` method allocates `an_x` and `an_y` on every call. With the default settings, this will allocate 800 bytes of RAM, reducing the number of other libraries that can be used. This is especially noticeable on the Arduino Uno, which only has 2KB of memory

I changed this to perform calculations (and modify) on two float arrays passed in the parameter. Although this goes against general programming paradigms, I believe this tradeoff is worth making to reduce memory usage.