aromring/MAX30102_by_RF #41: Can't Print SPO2 on OLED

AustinTetteh, 2024-07-10:
Thanks for your algorithm on spo2 which works better than the maxim's code. I'm trying to print the spo2 values on an OLED. For the OLED, I'm using ssd1306 asciiWire library which uses low memory. The spo2 algorithm works perfectly when I run it separately as I'm able to print the values on a serial monitor. The OLED also works perfectly when I display something on it. 

However when I merged the codes and try to display the spo2 on the OLED, the code resets and the setup function starts again.
The memory usage says sketch uses 40% of memory and dynamic memory uses 60%. 

I've used the maxim's algorithm and I get the values being displayed on the OLED. With that it, sketch uses 60% and dynamic memory uses 88%

AustinTetteh, 2024-07-10:
[mod_spo2_090925.zip](https://github.com/user-attachments/files/16168985/mod_spo2_090925.zip)
This is the merged code for the OLED and spo2 algorithm

aromring, 2024-07-12:
Is this a bug in my code? If yes, then submit all the data and graphics as described in README.
Other than that it could be some asciiWire conflict, but I'm wildly speculating.
Good luck debugging _your_ OLED code!

AustinTetteh, 2024-07-12:
These are the files
[Red and IR values.xlsx](https://github.com/user-attachments/files/16200024/Red.and.IR.values.xlsx)
[image]
Hoping to hear from you soon.
Regards.

AustinTetteh, 2024-07-13:
I've fixed it. It was memory issue. I think the Arduino didn't have enough memory to store 200 32 bit data samples. So I changed the sampling time to 2 seconds hence the buffer size became 50 and I recalculated the sum_X2 variable. It's working now.
Thanks

closed 2024-07-13 by AustinTetteh

aromring, 2024-07-18:
I am glad you've found the bug.