aromring/MAX30102_by_RF #42: How to address inaccurate heart rate caused by motion artifacts?

yuelsun, 2025-03-18:
Now I use Kalman filter to filter the data of max30102. I can detect heartbeat when measuring at the wrist in a static state. **However**, it becomes invalid during activity. I want to know what are the better algorithms that can eliminate motion artifacts? Thank you.

sergiocntr, 2025-03-22:
Hi everyone,

I noticed that a recent issue was closed because it was more about choosing the best algorithm for filtering data rather than a problem with the code itself. I believe the maintainer made the right decision since it's beyond the scope of the repository's direct concerns.

However, I think it could be very beneficial to leverage AI tools to explore this kind of question. Modern AI models can provide detailed insights into algorithm selection based on specific data characteristics, constraints, and performance considerations. Once a solid answer is formulated, the findings could then be shared back with the community, aligning with the open-source spirit of collaboration and knowledge sharing.

yuelsun, 2025-03-22:
Thanks for getting back to me, I'll optimize the algorithm through my own efforts.

sergiocntr, 2025-03-22:
> Thanks for getting back to me, I'll optimize the algorithm through my own efforts.

Hi,

What I managed to do wasn't satisfactory. There are multiple potential issues, including hardware problems (especially with fake modules) and errors in data acquisition. I experimented using Velcro strips and modified gloves to collect data from the sensor, but the results were not always reliable.

I also tested a low-cost oximeter, and even that proved to be quite sensitive. Ensuring a stable acquisition often requires mechanically securing the finger in place. Additionally, sensor placement and its relative position to the heart significantly affect readings due to variations in blood pressure. If you’re running and your arm is moving, I believe it will be quite challenging to get stable readings.

Another key point is that the device turns the detector on and off rather than keeping it continuously active. Also, for reliable results, it needs to sample data for at least 20 seconds in a stable position. If the finger moves—especially if its height relative to the heart changes—the process should reset and re-acquire data.

Lastly, I strongly recommend processing the acquired data in Python. It has powerful libraries for data analysis that microcontrollers like Arduino simply can’t match.

Best of luck, and I hope you succeed!

yuelsun, 2025-03-22:
Hello, I used Kalman filter, and I think the effect of detecting finger positions is quite good. In the static state, the wrist position can also be measured, but once it moves, it completely fails 😭. Currently, I set the sampling frequency of MAX30102 to 100Hz, and the sampling time is 4s. I found that if the sampling time is longer, the measured heart rate will be more accurate.

However, due to the complex structure of the wrist, there are a large number of motion artifacts. I tried filtering processing, but the effect was not ideal. In addition, I recently read some papers and found that more advanced filtering algorithms such as adaptive filters can make the data collection display relatively accurate heart rate waveform after filtering. I'm still trying to understand this algorithm.
Similarly, it can also be analyzed by deep learning in Python, but I'm sorry, my major limits me from spending more effort on learning deep learning.

Thank you again, and I hope you succeed.